dw-0/kiauh #652: installing klipper with the not python 3.8 variant fails on rpi like this

1knueller, 2025-02-24:
### Linux Distribution

rpi

### What happened

installing klipper failed

### What did you expect to happen

klipper should have been installed

### How to reproduce

start kiauh select 2 (i dont have python 3.8 configured on my machine)
install 1 klipper instance 

### Additional information

[✓ OK] Dependencies already met!

###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
Cloning into '/home/pi/klipper'...
remote: Enumerating objects: 71879, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (59/59), done.
error: RPC failed; curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

#=======================================================#
 Cloning Klipper from
 https://github.com/Klipper3d/klipper
 failed!
#=======================================================#

dw-0, 2025-02-24:
This is no direct issue with KIAUH. It could be as simple as a (temporary) network issues.

Matt2916, 2025-02-24:
I have the "same" issue with a slightly different error message:

Cloning into '/home/admin/klipper'...
remote: Enumerating objects: 71879, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (59/59), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 35612 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed

closed 2025-02-26 by dw-0