dw-0/kiauh #653: Unable to flash MKS Robin Nano V3 (STM32F407) with KIAUH v5.1.3 due to DFU signature warning treated as error

drackp2m, 2025-03-02:
### Linux Distribution

Debian GNU/Linux 11 (bullseye)
KIAUH v5.1.3
dfu-util 0.9

### What happened

After a recent update, I can no longer flash my MKS Robin Nano V3 board (STM32F407) using the "Regular" flash method in KIAUH. The process fails with the error message "No DFU capable USB device available" and exits prematurely.

The flashing process gets to the point where it shows:

```
dfu-util: Invalid DFU suffix signature dfu-util: A valid DFU suffix will be required in a future dfu-util release!!! dfu-util: No DFU capable USB device available
```

The error appears immediately after attempting to flash without giving the device a chance to enter DFU mode. Previously this process worked correctly, but now it fails consistently.

### What did you expect to happen

The flashing process should continue despite the "Invalid DFU suffix signature" warning message, as it did before. The MCU should be put into DFU mode automatically by the script, and the flash should complete successfully.

In previous successful flashes, although the same warning appeared, the process would continue and successfully flash the firmware, showing messages like:

```
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash"
Downloading to address = 0x0800c000, size = 32284
Download	[=========================] 100%        32284 bytes
Download done.
File downloaded successfully
```

### How to reproduce

1. Connect MKS Robin Nano V3 (STM32F407) board via USB
2. Run KIAUH and navigate to Advanced → Build and flash
3. Select the MCU (usb-Klipper_stm32f407xx_410029000A50325436303720-if00)
4. Choose "Regular" flash method
5. Select the firmware config file (in my case "super-racer.config") 
6. Confirm flash parameters
7. Observe that the process fails with the error:
   "Failed to flash to /dev/serial/by-id/usb-Klipper_stm32f407xx_410029000A50325436303720-if00: Error running dfu-util"
   "No DFU capable USB device available"

My MCU configuration:
- STM32F407
- 48KiB bootloader
- 8 MHz crystal
- USB (on PA11/PA12)

### Additional information

The key difference appears to be that now KIAUH treats the "Invalid DFU suffix signature" warning as a fatal error, while previously it recognized this as a non-critical warning and continued the flashing process.

I didn't need to press any buttons on the board for flashing before - the process was entirely automatic. This suggests that the automatic bootloader entry sequence is no longer working properly in the current version.

P.S.: This is issue has created with Cloud Sonet 3.7 thought Copilot Pro.

dw-0, 2025-03-13:
> After a recent update,

Update of what? A general system package update?

drackp2m, 2025-03-13:
> Did you do any system package updates with (or without - doesn't matter) KIAUH?

I always make a habit of updating system packages whenever I remember (they run through Mainsail at the same time as Klipper and Moonraker updates, etc.), but the problem is that I get an error in the Alpha version, but not in the stable version.

---

I'm currently printing a few things, but as soon as they're finished, I'll see if I remember to record a couple of videos so you can see the error / difference.

If there's a log I can pull up somewhere, don't hesitate to ask me for anything I can help with.

drackp2m, 2025-03-13:
> Advanced -> Build and Flash -> Regular flashing -> `make flash` -> USB -> Select MCU -> Flash

Yes, these are the steps for the stable version, with which I've never had any problems. I'm a fairly advanced user in this regard, although I explained myself a bit poorly in my first post (sorry). But I've been printing and using Klipper / KIAUH for many years, and of everything I've used in version 6 of the utility, it's the only one that has failed me.

And I recently used that version to install almost everything it includes on a new Raspberry Pi I purchased, and this flash step is the only one I haven't been able to complete successfully.

drackp2m, 2025-03-15:
I don't know if it changed anything, and the previous time I didn't really test all the options.

Today using the alpha version, it seems to work correctly:

- Select the flash method for flashing the MCU. => Regular flashing method.
- Which flash command to use for flashing the MCU? => make flash (default)
_*If I choose make serialflash (stm32flash), I get error, but my board is stm32. Maybe this is where my confusion came from the other day._
- How is the controller board connected to the host? => USB

However, trying to flash my Raspberry (as a secondary MCU), I do have problems. I choose the “Linux process” option under the “Micro-controller Architecture” option, and when it gets to the end of the process I see the following:

!!! NO FIRMWARE FILE FOUND !!!
Unable to find a compiled firmware file!

Make sure, that:
- the folder '~/klipper/out' and its content exist.
- the folder contains the following file:
    - 'klipper.elf'
    - 'klipper.elf.hex'

dw-0, 2025-03-15:
> However, trying to flash my Raspberry (as a secondary MCU), I do have problems.

Because that feature is not implemented yet. A feature request for this is already available here: #550 
The firmware you build is not going to be flashed to the Rpi.

Anyways, it seems like the initial issue is solved and there are no actions necessary on KIAUHs end.
Hence i will close this issue. Thanks for reporting back.

closed 2025-03-15 by dw-0

drackp2m, 2025-03-16:
Oh no, but I wasn't talking about the steps [explained here](https://www.klipper3d.org/RPi_microcontroller.html) to "activate" or configure it, but rather to update your Klipper version once it's working. In version 5 of KIAUH, it detects the newly prepared firmware (Linux process), while in v6, after compiling everything, it seems it's not stored in the directory where the script looks for the file to flash.

Anyway, I understand that this issue has nothing to do with the issue I initially opened, so I will open another one in the next few days when I gather more information about it (I would not like to open any additional false issues)