dw-0/kiauh #657: Check for dietpi critically fails on other distros

ulmeck, 2025-03-09:
### Linux Distribution

Rasberry PI OS (Debian Bookworm) (ARM)

### What happened

Installing as a dedicated, non-privileged user doing a manual git install. 
Klipper install fails immediately after checkout with

`[OK] Checkout successful!
[ERROR] Error during installation of Klipper requirements!
[ERROR] [Errno 13] Permission denied: '/boot/dietpi/.version'
[ERROR] Klipper installation failed!`

This is due to '/boot' not being world readable and fails in kiauh/components/klipper/klipper_setup.py

### What did you expect to happen

I would expect to be able to install without being a privleged user or being prompted for the sudo password. 

### How to reproduce

Install as a non-privileged user with Pi OS (Debian Bookworm).

dw-0, 2025-03-09:
Hi, thank you for reporting.
I pushed a fix for this. Please update and try again.

Keep in mind, that your user must have sudo permissions. Otherwise installing dependencies will fail.

dw-0, 2025-03-13:
fixed with #658