dw-0/kiauh #670: Support for low resouces / weird platforms

pavels, 2025-04-02:
### Is your feature request related to a problem? Please describe

I am working on Debian installation for K1

It would be useful to be able to

a) skip installation of `moonraker-speedups.txt` due to limitation of RAM on this platform (the compilation of these dependencies takes ages)
b) be able to define custom python version for klipper (i need to run klipper on python3.8 when the default system python is 3.11)

### Describe the solution you'd like

It would be fine to have this just in config - no need to clutter the UI. 
I am volunteering to create at - will you accept such PR?

dw-0, 2025-04-02:
Hi! Thanks for the request.
At first i was tempted to simply say: "Im not willing to accept a PR that's just for the sake of supporting any Creality platform/printer", but at second thought, i think having those two things may be useful in general and to other users as well.

So, what im thinking is, that we should add a visible config option for the moonraker speedup requirements in the default config like this maybe:
```diff
[moonraker]
++ # Moonraker supports two optional Python packages that can be used to reduce its CPU load
++ # If set to true, those packages will be installed during the Moonraker installation
++ optional_speedups: True

# add custom repositories here, if at least one is given, the first in the list will be used by default
# otherwise the official repository is used
#
# format: https://github.com/username/repository, branch
# example: https://github.com/Arksine/moonraker, master
#
# branch is optional, if given, it must be preceded by a comma, if not given, 'master' is used
repositories:
    https://github.com/Arksine/moonraker
```

Im not sure if we should add the python version config option already in the default config. My concern here is, that people may misinterpret the config option and may think kiauh will install those python versions along the way as long as there is no real documentation available (it's on my todo to write a kiauh documentation though). So for the simple reason of preventing accidental misconfiguration, we may want to omit adding those to the `default.kiauh.cfg` but allow them as (for now undocumented) custom config options like:
```ini
[klipper]
use_python: 3.8

[moonraker]
use_python: 3.9
```
If that option is not set, the default python should be used.

What are your thoughts?

pavels, 2025-04-02:
Hi

yes, that was my idea in the beginning - that although this would be useful for my specific usecase, it can be made generic enough that it makes sense to add it to kiauh.

speedups is clear and fine - this will work

python version - i understand the concern about the confusion.
Currently this is defined here https://github.com/dw-0/kiauh/blob/master/kiauh/utils/sys_utils.py#L109 as `/usr/bin/python3` so maybe what can also help remove the confusion, and make the code better is instead of configuring version, we can configure full path? so it can be for example `use_python_binary:`  (or anything similar) and the value would be `/usr/bin/python3.8` if you want 3.8 or even some strange path in your home if you use pyenv or something like that to manage python versions. That should also be more clear to users. If you are pointing it to particular binary, it is kind of a hint that the binary must exist in the first place. Again i completely agree that it would be done without entry in `default.kiauh.cfg`. If you think that just plain version is better, than that's fine as well - just it will require to do string interpolation to get the correct binary name to pass to virtualenv.

dw-0, 2025-04-02:
I like your suggestion of defining the binary path instead of defining only the version. Lets go with that :)
`use_python_binary` as the option name is fine, i like it and it describes what it does.

closed 2025-04-07 by pavels