dw-0/kiauh #674: Log Upload to Log Parser

CODeRUS, 2025-04-19:
### Is your feature request related to a problem? Please describe

n/a

### Describe the solution you'd like

I am wondering if you interested in uploading logs to my parser service:
https://coderus.openrepos.net/klipper_logs

Example log: https://coderus.openrepos.net/klipper_logs/15f8afeb74d2217c7bc52c91471b33e4

### Additional information

Sources of service:
https://github.com/CODeRUS/klipper_logs

dw-0, 2025-04-26:
Hi, sorry for answering so late.
I think it would be an interesting idea. Though, i would like to have it implemented as a second option for uploading logs.
Would you be interested in doing that implementation? The user could select the log and then select in a second dialog if he want's to upload it to you service?

CODeRUS, 2025-04-26:
Of course I can do it, I was wondering if there are any objections to uploading logs to a third party resource :)

dw-0, 2025-04-26:
Personally, i don't have any objections. :) Now that we are talking about, i forgot to migrate the upload agreement dialog from v5 and implement it in v6:
[image]

That way the user was very aware that the logs get uploaded to a third party service. If you will be already on it, would you mind migrating that dialog as well? That would be awesome and very much appreciated. We might need to adapt it a bit and either call it "will be uploaded to third party services" or list both services there.