dw-0/kiauh #676: Error with Python install when installing Klipper

s3anami, 2025-04-27:
### Linux Distribution

Debian 11
CB1 image from BTT
https://github.com/bigtreetech/CB1/releases/tag/V2.3.4

### What happened

The following error when installing klipper using KIAUH script

ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.9/py_compile.py", line 144, in compile
    code = loader.source_to_code(source_bytes, dfile or file,
  File "<frozen importlib._bootstrap_external>", line 853, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/compileall.py", line 238, in compile_file
    ok = py_compile.compile(fullname, cfile, dfile, True,
  File "/usr/lib/python3.9/py_compile.py", line 150, in compile
    raise py_exc
py_compile.PyCompileError:   File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 105, in _run_wrapper
    status = _inner_run()
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 96, in _inner_run
    return self.run(options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 68, in wrapper
    return func(self, options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 459, in run
    installed = install_given_reqs(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/__init__.py", line 83, in install_given_reqs
    requirement.install(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/req_install.py", line 867, in install
    install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 728, in install_wheel
    _install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 614, in _install_wheel
    success = compileall.compile_file(path, force=True, quiet=True)
  File "/usr/lib/python3.9/compileall.py", line 255, in compile_file
    msg = err.msg.encode(sys.stdout.encoding,
TypeError: encode() argument 'encoding' must be str, not None

### What did you expect to happen

Python to install correctly

### How to reproduce

Running the klipper install script reproduces the same issue with it is the new 6 release or stable 5 release

### Additional information

I am following the guide to move a Sovol SV08 to mainline klipper, the error occurs at step 4.4
Everything else is able to be installed correctly besides Python.

I have attempted reflashing the image to the EMMC and repeating install and the 6 version and 5 version and neither helps.

dezigenb, 2025-04-27:
I raised the same issue yesterday, but the author closed it for me as invalid.

dw-0, 2025-04-27:
Duplicate of #675 

I was able to reproduce the issue on a fresh debian 11 system. But to make it clear once again: This is NOT an issue with KIAUH. KIAUH expects/requires Klipper and other component to maintain their dependencies / requirements.

It looks like its caused by the outdated `python-can==3.3.4` requirement for Klipper's virtualenv.
That version of python-can seems to require python 2, but debian 11 (bullseye) has python 3.9 installed.
I will report this to the maintainer of Klipper, and ask if there is any way of updating those requirements.
I tested it by modifying the klipper-requirements.txt like mentioned here https://klipper.discourse.group/t/weird-python-errors-after-can-upgrade/19037/10 and changed `python-can==3.3.4` to `python-can~=4.3`. Installation succeeded afterwards.

## Before:
[image]

## After:
[image]

xtalkme, 2025-04-27:
I also had some luck with the following steps:

rm -rf ~/.cache 
~/klippy-env/bin/pip3 install pyserial aenum wrapt pycparser markupsafe greenlet python-can Jinja2 cffi 
~/kiauh/kiauh.sh

When running kiauh.sh install klipper, do all the defaults BUT do NOT remove the virtual environment.

dw-0, 2025-04-27:
> do all the defaults BUT do NOT remove the virtual environment

This is only a "solution" if it's *not* a fresh installation of Klipper though.

GcodeGearHead, 2025-04-27:
> I also had some luck with the following steps:
> 
> rm -rf ~/.cache ~/klippy-env/bin/pip3 install pyserial aenum wrapt pycparser markupsafe greenlet python-can Jinja2 cffi ~/kiauh/kiauh.sh
> 
> When running kiauh.sh install klipper, do all the defaults BUT do NOT remove the virtual environment.

this solved the issue for me

dezigenb, 2025-04-28:
> Duplicate of [#675](https://github.com/dw-0/kiauh/issues/675)
>
> I was able to reproduce the issue on a fresh debian 11 system. But to make it clear once again: This is NOT an issue with KIAUH. KIAUH [...]

It works, thank you for your help.

dw-0, 2025-05-01:
@docktermj Thanks for the info!
Im still not sure how to tackle this. I talked to the Klipper maintainer and presumably this is not a Klipper issue (but not a KIAUH one as well, as we see). It's really weird, because Klipper was able to be installed on Debian 11 systems without any issues pretty much when Debian 11 was released. We are talking about years here, where there were no real issues of installing Klipper on an OS which had only Python 3 installed.

I did some testing and i was able to successfully install Klipper on Debian 12 (well, kinda - Debian 12 causes it's own issues apparently, see #678) without the need of modifying the klippy-requirements.txt.

For now, im not sure where to start troubleshooting and cannot explain why suddenly the python-can dependency and its dependency of aenum seems to cause issues.

rogerlz, 2025-05-01:
FYI `aenum` released a new version a few days ago: https://pypi.org/project/aenum/#history and `python-can` 3.3.4 does not specify the `aenum` version: https://github.com/hardbyte/python-can/blob/release-3.3.4/setup.py#L115

KevinOConnor, 2025-05-01:
Thanks.  So, does this change to Klipper fix the issue?
```diff
--- a/scripts/klippy-requirements.txt
+++ b/scripts/klippy-requirements.txt
@@ -7,6 +7,7 @@ pyserial==3.4
 greenlet==2.0.2 ; python_version < '3.12'
 greenlet==3.0.3 ; python_version >= '3.12'
 Jinja2==2.11.3
+markupsafe==1.1.1       # Needed by Jinja2
 python-can==3.3.4
-markupsafe==1.1.1
+aenum==3.1.15           # Needed by python-can
 setuptools==75.6.0 ; python_version >= '3.12' # Needed by python-can
```

-Kevin

dw-0, 2025-05-02:
Hi @KevinOConnor , sadly this does not fix the issue, please see the output here:
```
###### Set up Python virtual environment ...
created virtual environment CPython3.9.2.final.0-64 in 122ms
  creator CPython3Posix(dest=/home/dw/klippy-env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/dw/.local/share/virtualenv)
    added seed packages: pip==20.3.4, pkg_resources==0.0.0, setuptools==44.1.1, wheel==0.34.2
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
[OK] Setup of virtualenv successful!

###### Updating pip ...
Requirement already satisfied: pip in ./klippy-env/lib/python3.9/site-packages (20.3.4)
Collecting pip
  Downloading pip-25.1-py3-none-any.whl (1.8 MB)
     |████████████████████████████████| 1.8 MB 2.2 MB/s
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed pip-25.1
[OK] Updating pip successful!

###### Installing Python requirements ...
Ignoring greenlet: markers 'python_version >= "3.12"' don't match your environment
Ignoring setuptools: markers 'python_version >= "3.12"' don't match your environment
Collecting cffi==1.14.6 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 5))
  Downloading cffi-1.14.6-cp39-cp39-manylinux1_x86_64.whl.metadata (1.2 kB)
Collecting pyserial==3.4 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 6))
  Downloading pyserial-3.4-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting greenlet==2.0.2 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 7))
  Downloading greenlet-2.0.2-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (4.1 kB)
Collecting Jinja2==2.11.3 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 9))
  Downloading Jinja2-2.11.3-py2.py3-none-any.whl.metadata (3.5 kB)
Collecting python-can==3.3.4 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 10))
  Downloading python-can-3.3.4.tar.gz (179 kB)
  Preparing metadata (setup.py) ... done
Collecting aenum==3.1.15 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 11))
  Downloading aenum-3.1.15-py3-none-any.whl.metadata (3.7 kB)
Collecting markupsafe==1.1.1 (from -r /home/dw/klipper/scripts/klippy-requirements.txt (line 12))
  Downloading MarkupSafe-1.1.1-cp39-cp39-manylinux2010_x86_64.whl.metadata (3.2 kB)
Collecting pycparser (from cffi==1.14.6->-r /home/dw/klipper/scripts/klippy-requirements.txt (line 5))
  Downloading pycparser-2.22-py3-none-any.whl.metadata (943 bytes)
Collecting wrapt~=1.10 (from python-can==3.3.4->-r /home/dw/klipper/scripts/klippy-requirements.txt (line 10))
  Downloading wrapt-1.17.2-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (6.4 kB)
Downloading cffi-1.14.6-cp39-cp39-manylinux1_x86_64.whl (405 kB)
Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
Downloading greenlet-2.0.2-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (610 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 610.9/610.9 kB 3.4 MB/s eta 0:00:00
Downloading Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Downloading aenum-3.1.15-py3-none-any.whl (137 kB)
Downloading MarkupSafe-1.1.1-cp39-cp39-manylinux2010_x86_64.whl (32 kB)
Downloading wrapt-1.17.2-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (82 kB)
Downloading pycparser-2.22-py3-none-any.whl (117 kB)
Building wheels for collected packages: python-can
  Building wheel for python-can (setup.py) ... done
  Created wheel for python-can: filename=python_can-3.3.4-py2.py3-none-any.whl size=154190 sha256=922acad88d5c0ad5fc4ed1cecd8e0b18ba4bb40b4c9d2922247a3506542df381
  Stored in directory: /home/dw/.cache/pip/wheels/af/c4/49/95c8dc63f382ce9a822c5dd73b8b1215e63cb5d001c75a87a5
Successfully built python-can
Installing collected packages: pyserial, aenum, wrapt, pycparser, markupsafe, greenlet, python-can, Jinja2, cffi
  DEPRECATION: Building 'python-can' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'python-can'. Discussion can be found at https://github.com/pypa/pip/issues/6334
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.9/py_compile.py", line 144, in compile
    code = loader.source_to_code(source_bytes, dfile or file,
  File "<frozen importlib._bootstrap_external>", line 853, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/dw/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
[... 97 lines cut ...]
  Downloading typing_extensions-4.13.2-py3-none-any.whl.metadata (3.0 kB)
Collecting packaging (from python-can==4.1->-r /home/dw/klipper/scripts/klippy-requirements.txt (line 10))
  Downloading packaging-25.0-py3-none-any.whl.metadata (3.3 kB)
Collecting msgpack~=1.0.0 (from python-can==4.1->-r /home/dw/klipper/scripts/klippy-requirements.txt (line 10))
[... cut ...]
Installing collected packages: pyserial, wrapt, typing-extensions, pycparser, packaging, msgpack, markupsafe, greenlet, python-can, Jinja2, cffi
Successfully installed Jinja2-2.11.3 cffi-1.14.6 greenlet-2.0.2 markupsafe-1.1.1 msgpack-1.0.8 packaging-25.0 pycparser-2.22 pyserial-3.4 python-can-4.1.0 typing-extensions-4.13.2 wrapt-1.17.2
[OK] Installing Python requirements successful!
```

It seems its even solving the deprecation warning caused by pip <25.0, which was reported in #678 

Would Klipper work fine, with upgrading to a new major version of python-can?

dw-0, 2025-05-02:
I pushed a fix. It seems like, as @KevinOConnor pointed out, the actual issue was KIAUH always updating pip before it installs any packages. This caused this issue, and issue #678. Both of them should be fixed with that change, at least i did not encounter the issues anymore when i tried reproducing the issue. I keep this issue open for now so i can maybe get some more confirmations that the issue is gone.

GcodeGearHead, 2025-05-04:
I just installed Klipper today on a fresh machine. 
Used the latest KIAUH install script and can confirm no errors.

dw-0, 2025-05-05:
Thanks for the feedback @GcodeGearHead 
In that case, i will close this issue for now.

closed 2025-05-05 by dw-0

dezigenb, 2026-02-05:
> I am having the greenlet issue now again, actually :/
> 
> WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/greenlet/ WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/greenlet/ WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/greenlet/ WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/greenlet/ WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/greenlet/ ERROR: Could not find a version that satisfies the requirement greenlet==3.1.1 (from versions: none) ERROR: No matching distribution found for greenlet==3.1.1
> 
> [ERROR] Error installing Python requirements: Installing Python requirements failed! [ERROR] Error during installation of Klipper requirements! [ERROR] Error installing Python requirements: Installing Python requirements failed! [ERROR] Klipper installation failed!
> 
> What to do?
> 
> Do I make a new post?
> 
> Or is there one already? I have not been able to find one talking about this lately?

This seems like a problem with your network. Try checking your network and then try again

LukasExemplar, 2026-02-06:
I made a new issue post instead. It doesn't seem to be an issue with my network, I've tested that already. But maybe I am wrong.