dw-0/kiauh #678: Building using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version

csecsoliver, 2025-04-29:
### Linux Distribution

Debian 12 bookworm

While installing klipper on a clean debian 12 install in virtualbox, this error / warning showed up and kiauh detected it as a fatal error.

### What happened

While installing klipper on a clean debian in virtualbox, this error / warning showed up and kiauh detected it as a fatal error.

### What did you expect to happen

I have no great insight into what this error means, but it might be ok to continue without making the whole process fail. However in the future this will have to be solved elsewhere.

### How to reproduce

apt update
apt install git
download and run kiauh using the readme.md simple instructions
select to use version 6
select install
select klipper
use all default options
wait

csecsoliver, 2025-04-29:
The install didn't fail and reinstalling it without overwriting seems to have bypassed the issue (further observations necessary)

dw-0, 2025-04-30:
I was able to reproduce the issue. The actual deprecation warnings were this:
```sh
DEPRECATION: Building 'cffi' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'cffi'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  DEPRECATION: Building 'python-can' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'python-can'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  DEPRECATION: Building 'markupsafe' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'markupsafe'. Discussion can be found at https://github.com/pypa/pip/issues/6334
```

Workaround: run the installation a second time so the remaining setup steps are executed.

That's obviously not a graceful solution, as much as the command exiting with a status code of != 0 is, causing the setup to fail, although setting up the virtualenv worked.

I have to do some research on how to handle this.

closed 2025-05-04 by dw-0