dw-0/kiauh #681: Ports issues

matthewjhinkleman, 2025-05-04:
### Linux Distribution

Raspberry pi os bullseye & bookworm

### What happened

I installed klipper, Moonraker, and mainsail, and the web interface would not load unless it was on a different port than 80

### What did you expect to happen

The web interface to load

### How to reproduce

Raspberry pi 4. Follow instructions in the read me.

dw-0, 2025-05-04:
The provided information is not enough for proper troubleshooting.
Check if there is any other application running on port 80. Chances are, that this is currently the case.

matthewjhinkleman, 2025-05-09:
I checked and something (I forget what) was conflicting on port 80. I used only kiauh on a fresh install of rapsberry pi os lite so is that something that can be fixed in the script?

matthewjhinkleman, 2025-05-10:
How would I do that? I have already spent hours trying to fix it.

closed 2025-05-31 by dw-0