dw-0/kiauh #682: How do I add projects to Kiauh?

BlackNet, 2025-05-07:
### Is your feature request related to a problem? Please describe

TLDR There isa  project I would like to add to Kiauh.  I wanted to know how to go about adding that.

### Describe the solution you'd like

Add existing projects.

dw-0, 2025-05-10:
Hi, i haven't found the time to document the actual steps required but it's not really hard. Installers can be added via a community extension. A rough explanation of that idea can be found here: https://github.com/dw-0/kiauh/issues/330#issuecomment-1934932795

Some implementations that could be used as reference are octoapp or klipper_backup:
https://github.com/dw-0/kiauh/blob/master/kiauh/extensions/octoapp/octoapp_extension.py
https://github.com/dw-0/kiauh/blob/master/kiauh/extensions/klipper_backup/klipper_backup_extension.py

A python class that inherits from the base class and a proper metadata.json are enough:
https://github.com/dw-0/kiauh/blob/master/kiauh/extensions/base_extension.py

I hope you can work with those information. Feels free to ask specific questions in case something is unclear.

BlackNet, 2025-05-21:
I am sorry for the late reply.  Thank you for the info on this. I will look into what needs to be done to add some new projects.