dw-0/kiauh #684: Unable to update klipper and moonraker

diega, 2025-05-18:
### Linux Distribution

$ uname -a
Linux klipper 6.12.20+rpt-rpi-v7 #1 SMP Raspbian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) armv7l GNU/Linux
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### What happened

I was trying to update my installation using the built in upgrade option
  * Klipper: v0.13.0-51 -> v0.13.0-91
  * Moonraker: v0.9.3-72 -> v0.9.3-74
 
and I got this error (red) message:
```
[notice] A new release of pip is available: 25.0.1 -> 25.1.1
[notice] To update, run: python -m pip install --upgrade pip

[ERROR] Error installing Python requirements: Installing Python requirements failed!
[ERROR] An unexpected error occured:
Error installing Python requirements: Installing Python requirements failed!
Traceback (most recent call last):
  File "/home/diegoll/kiauh/kiauh/utils/sys_utils.py", line 198, in install_python_requirements
    raise VenvCreationFailedException("Installing Python requirements failed!")
utils.sys_utils.VenvCreationFailedException: Installing Python requirements failed!

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/diegoll/kiauh/kiauh/core/menus/base_menu.py", line 229, in run
    selected_option.method(
  File "/home/diegoll/kiauh/kiauh/core/menus/update_menu.py", line 201, in update_moonraker
    self._run_update_routine("moonraker", mrsvc.update)
  File "/home/diegoll/kiauh/kiauh/core/menus/update_menu.py", line 307, in _run_update_routine
    update_fn(*args)
  File "/home/diegoll/kiauh/kiauh/components/moonraker/services/moonraker_setup_service.py", line 186, in update
    install_python_requirements(MOONRAKER_ENV_DIR, MOONRAKER_REQ_FILE)
  File "/home/diegoll/kiauh/kiauh/utils/sys_utils.py", line 205, in install_python_requirements
    raise VenvCreationFailedException(log)
utils.sys_utils.VenvCreationFailedException: Error installing Python requirements: Installing Python requirements failed!
```

Both packages failed with similar errors

### What did you expect to happen

to properly install the new versions

### How to reproduce

  * `$ ./kiauh/kiauh.sh`
  * Perform action **2**
  * Select **1** or **2** to update Klipper or Moonraker

dw-0, 2025-05-18:
Please try to remove and re-install klipper and moonraker. I assume pip 25 may cause some issues. When uninstalling klipper and moonraker, make sure to also remove the python environment.

Please report back if this solves your issue or not. If it does not, i need to inspect further.

diega, 2025-05-18:
Is there any chance to leave my system unusable if I do that? right now, even with the failing update it's still working (with the previous version)

dw-0, 2025-05-18:
In case the dependency installation fails again, yes. I would suggest you do a backup of klipper and moonraker before trying to re-install. Kiauh is able to do that for you. That way, in case of a failure, you can copy the python environment folders from the backup folder to the original destination.

diega, 2025-05-18:
I removed and re-installed both, moonraker and klipper, and both installed just fine (I had to retry the installation a few times b/c some "Remote end closed connection without response" errors but I think that's beyond the scope of this issue). I don't know what could have been the problem with the initial update attempt. Feel free to close this issue if you think this is a fair workaround, I can confirm it worked.