dw-0/kiauh #687: Error after installing python requirements with retry and return code = 0

dw-0, 2025-05-31:
> Further testing shows this output:
> 
>  ```
>  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pillow/pillow-11.1.0-cp311-cp311-linux_armv7l.whl
>   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/zeroconf/zeroconf-0.131.0-cp311-cp311-manylinux_2_36_armv7l.whl
>   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/dbus-fast/dbus_fast-2.28.0-cp311-cp311-manylinux_2_36_armv7l.whl
> 
> Return code 0
> [ERROR] Error installing Python requirements: Installing Python requirements failed!
> [ERROR] Error during installation of Moonraker requirements!
> [ERROR] Error while installing Moonraker: Error installing Python requirements: Installing Python requirements failed!
> ```
> 
> So, even though the result.returncode is 0, the installation procedure fails because there are warning messages of result.stderr.  It should not fail if result.returncode is 0.
> 
> Suggest changing `if result.returncode != 0 or result.stderr:` to `if result.returncode != 0:` 

 _Originally posted by @DeanCording in [#595](https://github.com/dw-0/kiauh/issues/595#issuecomment-2925269210)_

Hey @DeanCording, thank you for the hint. Can you confirm that Moonraker is working correctly even after the error from KIAUH? Did all requirements got installed after the (i count 3) retries, so that the installation of those was actually successfull?

dw-0, 2025-05-31:
> What tipped me off was that the packages that received the warnings were changing between runs.

That's indeed kinda odd. But i have no explanation for it.
Your suggestion sounds about right though, that the condition should be changed and checking for `result.stderr` should be removed, as it isn't reliable in this case.

DeanCording, 2025-05-31:
I think it is just random network errors or server load issues.