dw-0/kiauh #713: fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com

KelC25, 2025-08-09:
### Linux Distribution

I am attempting to install coprint KCM Set on my Elegoo Neptune 4 3D printer, I have followed the instructions to enter the various commands on Windows as follows:

- ssh mks@169.254.224.149
- enter password
- C:\Users\info>ssh mks@169.254.224.149
mks@169.254.224.149's password:
           _              _
 _ __ ___ | | _____ _ __ (_)
| '_ ` _ \| |/ / __| '_ \| |
| | | | | |   <\__ \ |_) | |
|_| |_| |_|_|\_\___/ .__/|_|
                   |_|
Welcome to Armbian 22.05.0-trunk  with bleeding edge Linux 5.16.20-rockchip64

No end-user support: built from trunk

System load:   6%               Up time:       46 min
Memory usage:  18% of 976M      IP:            169.254.224.149
CPU temp:      45°C             Usage of /:    66% of 6.6G
storage/:      4% of 7.5G

Last login: Thu Nov 23 00:08:41 2023 from 169.254.139.2
mks@mkspi:~$ ./kiauh/kiauh.sh

**fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com**
/=======================================================\
|              New KIAUH update available!              |
|-------------------------------------------------------|
|  View Changelog: https://git.io/JnmlX                 |
|                                                       |
|  It is recommended to keep KIAUH up to date. Updates  |
|  usually contain bugfixes, important changes or new   |
|  features. Please consider updating!                  |
\=======================================================/
###### Do you want to update now? (Y/n):

### What happened

fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com and whilst I can remove Kipper, Monraker, Fluidd etc. I cannot install or update

### What did you expect to happen

I expected to be able to update my klipper software, remiove and reinstall Klipper and Moonraker, remove Fluidd and Install Mainsail

### How to reproduce

As per above

KelC25, 2025-08-09:
I have checked under my Internet and Network settings and I do not have a proxy configured, is this required?

dw-0, 2025-08-10:
I don't know which instructions you followed. In any case, all of this is not related to this project.
For general issues with the system, or issues when following any 3rd party instructions, please consult the corresponding sources.