dw-0/kiauh #724: Segmentation fault

tehniemer, 2025-09-20:
### Linux Distribution

`Raspberry Pi OS Lite (64bit)`
and
`MainsailOS 2.2.0`

### What happened

I changed the kipper repo to kalico and after the install invoking `./kaiuh/kaiuh.sh` hangs and eventually states `Segmentation fault` and kicks back to the command prompt. This happened twice, both fresh installs.

### What did you expect to happen

kaiuh to work.

### How to reproduce

install kaiuh
change to kalico main
try to run kaiuh again

dw-0, 2025-09-20:
> ### How to reproduce
>install kaiuh
>change to kalico main
>try to run kaiuh again

Hi, i wasn't able to reproduce the issue with the provided reproduction steps.
Also, im not quite sure what kind of error you actually get. A screenshot would be nice. I don't think it is really a KIAUH induced error. A segmentation fault is usually a memory access violation. Currently i cannot imagine, how something like that could be caused by KIAUH.

tehniemer, 2025-09-20:
[image]

[image]

dw-0, 2025-09-20:
I see. The issue is, that you modified that file manually. Please read the 5th line in that file. There is a reason for that.

For v5, you need to create a `klipper_repos.txt` file inside of the KIAUH folder and add the repos like in this example file: https://github.com/dw-0/kiauh/blob/v5/klipper_repos.txt.example

I suggest to use KIAUH v6 instead, it has a better integration of that feature by now and you don't need to manually fiddle around with textfiles.

Closing

tehniemer, 2025-09-20:
I didn't modify the file manually, I added the repo as instructed by kaiuh by

` echo "https://github.com/KalicoCrew/kalico, main" >> /home/pi/kiauh/klipper_repos.txt`

The only manual modification I did was comment out the bottom two lines to get it to work again. Please re-open.

dw-0, 2025-09-20:
The whitespace between the repo and the branch name isn't allowed. You need to remove that.
To fix your issue, you NOW need to modify the .kiauh.ini and remove that whitespace there as well:
```diff
-- custom_klipper_repo_branch= main
++ custom_klipper_repo_branch=main
```